daisyaudio/DaisySP #2: Folder typo in build instructions

SSteve, 2020-03-06:
The [build instructions for Mac](https://github.com/electro-smith/DaisySP/wiki/1.-How-to-Build#Mac) say

> Once downloaded, extract the folder from the .tar file, and move it to a destination of your choice for example, `~/Documents/`

I think that should be `~/Developer/` (like it is in the Linux instructions) because the environment variable added to .bash_profile is `ARMGCCPATH=~/Developer/gcc-arm-none-eabi-9-2019-q4-major/bin`

stephenhensley, 2020-03-06:
Funny, we actually had it that way first!

We just went through the steps on a Mac with nothing previously installed. There is no Developer folder by default. So we figured for those who don't have a bunch of experience, it would be easier to put it somewhere that exists on every Mac.

SSteve, 2020-03-06:
Changing the ARMGCCPATH works too. 

Dang! I missed making the first issue by three minutes!

stephenhensley, 2020-03-06:
@SSteve  Cool, I updated the wiki to fix ARMGCCPATH. I'll close this now that they're consistent.

If it makes you feel better, I only copied the first issue over from the old shared repo. So yours was the first real issue!

closed 2020-03-06 by stephenhensley