daisyaudio/DaisySP #3: Errors when building examples

SSteve, 2020-03-06:
I just set up the development environment under macOS 10.14.6. When I run `./rebuild_all.sh` building DaisySP seems to work but there are errors when building the examples. Here's the output:

```Rebuilding Everything. . .
Building DaisySP
Done.
./rebuild_examples.sh: line 2: LIBDAISY: command not found
rebuilding examples:
rebuilding examples/adenv/
Makefile:14: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:14: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/bypass/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/compressor/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/crossfade/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/dcblock/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/decimator/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/delayline/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/line/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/metro/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/nlfilt/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/oscillator/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/phasor/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/pitchshifter/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/pluck/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/port/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/reverbsc/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/svf/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/template/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/tone/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
rebuilding examples/whitenoise/
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
Makefile:13: /core/Makefile: No such file or directory
make: *** No rule to make target `/core/Makefile'.  Stop.
done
creating daisysp markdown files. . . 
    PolyPluck
No example Provided for PolyPluck
    adenv
    compressor
    crossfade
    dcblock
    decimator
    delayline
    dsp
No example Provided for dsp
    limiter
No example Provided for limiter
    line
    metro
    mode
No example Provided for mode
    nlfilt
    oscillator
    phasor
    pitchshifter
    pluck
    port
    reverbsc
    svf
    tone
    whitenoise
done.
Generating documenation for  doc/daisysp_reference.pdf
Created doc/daisysp_reference.pdf
done.
Creating Style Guide PDF
finished.
```

SSteve, 2020-03-06:
I just looked at rebuild_examples.sh. This line looks like a problem:

`LIBDAISY ?= C:/Users/shensley/Documents/Mothership/electrosmith/libdaisy`

stephenhensley, 2020-03-06:
@SSteve the examples rely on the other Daisy library, libDaisy, in order to access the hardware. So a LIBDAISY environment variable pointing to the location of the other library.

We're still finalizing a lot of details there, and there are many moving pieces still. In about a week or two we'll open that library up to. Until then, unfortunately the examples cannot be built.

Sorry for the inconvenience!

SSteve, 2020-03-07:
No worries. I'm happy to be along for the spin-up ride. It's not like I have any hardware to run it on yet, anyway! But I'm backer number 71 and I'm looking forward to having a Daisy to play with.

stephenhensley, 2020-03-07:
@SSteve you may be interested in what I just started working on with PR #5 -- Might try to get some example with portaudio or RtAudio working this weekend.

stephenhensley, 2020-05-29:
@SSteve after a long wait, we can finally close this issues

see: https://github.com/electro-smith/DaisyWiki/wiki for a Getting started guide.

closed 2020-05-29 by stephenhensley