daisyaudio/DaisySP #6: Add a .vcxproj in every example directory

touil, 2020-03-09:
Currently, the main Visual Studio solution/project compiles fine using the VisualGDB addin. It would be nice to have the same for the examples. Having a project template would be even better!

stephenhensley, 2020-03-09:
good idea! It'd be nice to have each example vcxproj in the main solution as well.

The bypass/template examples are good starting points/templates. I use a simple python script to duplicate projects, and do a little text replace for starting new projects. I can add that to misc/ with a few of the other scripts in there.

I've seen the ability to start a project from a template from VisualStudio, but not exactly how to make a template.

stephenhensley, 2020-12-23:
9 months later, we finally have this! 😄 

The DaisyExample repo now has VisualGDB and VS Code debug support for all examples.

In addition, I will be migrating all DaisySP examples to the DaisyExamples/seed folder to be more consistent with the hierarchy of the Hardware-based examples, and allow them to benefit from the new helper scripts.

As for a template, there are files availabe in  DaisyExamples/util, or you can use the new `helper.py` script to create a new project for any of the supported breakout boards by running:

`./helper.py create seed/MyNewProject` 

You can also specify a breakout board, but it creates a new daisy seed project by default.

closed 2020-12-23 by stephenhensley