daisyaudio/DaisySP #7: Install instructions for dependencies on macOS are incomplete

sedurCode, 2020-03-27:
Hi, great work so far! 

Unfortunately, installing the ARM toolchain as per the instructions on the Wiki on macOS Catalina did not work for me as macOS security 'cannot confirm the developer identity' of the compiler. Of course, there are workarounds and homebrew casks etc for getting this done. 

I'm sure this has been discussed before (and perhaps I have missed it), as per the mutable build environment, would it be appropriate to have a contained build environment for Daisy? If so, you could hook up something like VSCode to behave as a cross-platform IDE which operates on shared files between the container and the host OS. Further use of openOCD and thus GDB means you would have a consolidated workflow for code>make>debug that uses the same tools across all platforms. Of course, there are many benefits and drawbacks to this approach.

stephenhensley, 2020-03-31:
@sedurCode Thanks for the feedback! I've been testing the mac builds on macOS Sierra (10.12.6), and I don't recall experiencing the security issue. I'll look into this and see if I can do a fresh install on a system with Catalina. Once I do I can update the wiki.

At this time, the brew versions of the arm-toolchain are unsupported (I think they are dated a bit before the processor was fully supported, but it may work. I've only done attempted the build once or twice using the brew installed toolchain.

A contained build environment is a good idea for sure, and has been discussed a little so far.

stephenhensley, 2020-05-29:
@sedurCode the OS X toolchain installation instructions on the [DaisyWiki page](https://github.com/electro-smith/DaisyWiki/wiki) have been updated, and should be complete (including the security settings necessary for OS X Catalina).

We also found a brew formula that works for the toolchain. So getting started on Mac is shockingly fast now.

There are some gotchas with openocd, but that's only necessary if you want to debug with JTAG.

closed 2020-05-29 by stephenhensley

sedurCode, 2020-05-31:
looks like instructions worked flawlessly! 
Thanks for the amazing work @stephenhensley !!!

stephenhensley, 2020-06-01:
@sedurCode glad we were able to help! Credit for the thorough Mac OS X instruction belong to @jrepp

Included there are details for building a compatible version of openocd (as there the current binary releases have some issues with support for tthe MCU at this time).

As for debugging, I personally use VisualStudio with VisualGDB (not sure if either/both of those are available on Mac nowadays).

I've been meaning to try out STM32CubeIDE, but I did see a few people mentioning installation issues on Mac OS X 

When installing some of STs other Java software, I've found that you have to actually open the App package, and double click the 'setup' unix executable or run it from a shell to get it to install correctly, though I have not tried to setup the CubeIDE yet. This was from my experience installing STM32CubeProg, but I suspect the issues may be similar.

Depending on your familiarity/comfort with command line tools you _can_ get by with just openocd and gdb (or a simple gui for gdb like ddd), but that is by no means easy.

sedurCode, 2020-06-01:
Thanks @stephenhensley @jrepp 
I've been using STM32CubeIDE with an F4 disco board on both PC and Mac (Catalina) perfectly well, 
no problems at all so far. I can't remember if I had to do anything special when installing STM32CubeIDE on Mac, so I would suspect I did not. I think because STM32CubeIDE is another modified version of Eclipse it gets the benefit of more widespread support and development that is Mac friendly. I would definitely recommend giving it a punt as it might give a fairly complete install option for toolchain, editor and debugger all in one for those who are not so command-line savvy. 

I have bought a copy of VisualGDB for my work on PC, but I am quite sure it isn't on Mac. 
When my stuff is delivered I will also give VisualGDB a go, thank you :)