daisyaudio/DaisySP #28: Module subfolders

TheSlowGrowth, 2020-05-29:
A suggestion: Mode modules into subfolders to make them easier to find.
I would assume that this library can grow rather quickly and it might be a good idea to organize things from the start.
Subfolders could include:

- modulators
- soundsources
- utilities
- filters
- fx
- ... more?

stephenhensley, 2020-05-29:
I like this idea. I think it's still small enough that this isn't entirely necessary yet.

However, as you said, this library could grow quite quickly, and at that point something like this would really help finding the things you're looking for.

TheSlowGrowth, 2020-05-30:
Yes that was exactly my thought. Right now, things are still small enough that such a change wouldn't hurt too much. Should I make a pull request and suggest a folder structure?

stephenhensley, 2020-05-30:
@TheSlowGrowth Sure! That sounds great.

andrewikenberry, 2021-01-15:
Initial Folder Outline:

- Filters
- Effects
- Synthesis
- Physical Modeling
- Utility
- Dynamics
- Noise
- Drum
- Control

closed 2021-02-05 by stephenhensley