daisyaudio/DaisySP #31: add .clang-format file for automatic code formatting

TheSlowGrowth, 2020-05-30:
To help enforce the coding style, it could be helpful to add a .clang-format file at the repository root. In Visual Studio, a simple keycombination can then be used to directly reformat portions of code - or an entire file.

Extending on this: Pull request can be tied to an automatic formatting check supplied by clang-format. I can help with that.

stephenhensley, 2020-05-30:
oh -- we actually use a clang-format file. I thought it was in this repo, but I think its in the DaisyExamples repo and I hadn't noticed.

stephenhensley, 2020-05-30:
@TheSlowGrowth I was hoping to start setting up some CI and style checking via pull requests this weekend. I'll open an issue for us to track that separately from here. 

I really like the idea of checking style at the PR stage. As that cuts out a large amount of (valuable) review time.

commit 87a06da2 (2020-05-30): Merge pull request #34 from electro-smith/adding_clangformat

fixed #31 - added missing .clang-format
commit e0d778fe (2020-05-30): fixed #31 - added missing .clang-format
closed 2020-05-30 by stephenhensley