daisyaudio/DaisySP #35: Add automated style checks from PR

stephenhensley, 2020-05-30:
@TheSlowGrowth mentioned setting up clang-format checks from PRs

This would be immensely useful, and time-saving.

I've been using the clang-format stuff with Visual Studio for a while. So I'm pretty much sold on using that on git for the style checks. But I've also seen stuff like AStyle used for this.

Open to other alternatives as well, but clang-format feels like the frontrunner.

TheSlowGrowth, 2020-05-30:
There could be multiple checks. I'm currently trying out travis CI for running doxygen (see #29). That could be another neat check to ensure a consistent documentation.

👍 for clang-format, its very powerful and a great help during development!

TheSlowGrowth, 2020-05-30:
Seems to work now, but the code needs a bunch of reformatting at the moment.

stephenhensley, 2020-05-30:
As I mentioned in #29 I'm going to get the github.io page setup this weekend. So we'll be able to start working on the deployment at that point. Should be a bit later tonight.

TheSlowGrowth, 2020-05-30:
The jobs seem to work, though the documentation job takes forever. The problem is that latex has to be installed each time which is a huge pain. Looking for better solutions, but it seems complicated.

We could skip pdf generation for doxygen but the code style document still needs it.

TheSlowGrowth, 2020-05-30:
seems like I fixed it

stephenhensley, 2020-05-30:
oh cool. I was going to say we could take out the pdf stuff, and just get the Style-Guide moved to doxygen sooner than later. But it would be nice to be able to distribute pdf documentation.

stephenhensley, 2020-05-31:
Deployed

closed 2020-05-31 by stephenhensley