daisyaudio/DaisySP #43: Doc Deployment is failing

stephenhensley, 2020-06-01:
There are a few warnings that I don't see when building the docs on my end, but what's weird is it worked before we added the conditional.

I think its the lack of skip cleanup as it is running `git stash --all` before deploying.

TheSlowGrowth, 2020-06-01:
I had removed it due to warnings in the Travis GUI. It told me that this command is ignored or something like that. Oops.

stephenhensley, 2020-06-01:
@TheSlowGrowth yeah, I saw the same warning, but weirdly without it the build output still suggests using it.

the warning seems to suggest using `cleanup` instead. I can look into this a bit more later on.

stephenhensley, 2020-06-04:
doc deployment is succeeding. I'm going to set up CI for libdaisy and daisy examples later this week. So I'll revisit the `cleanup` vs `skip_cleanup` stuff while I'm doing that.

closed 2020-06-04 by stephenhensley