daisyaudio/DaisySP #59: Process single sample or block, return or replace inconsistency

AvAars, 2020-07-14:
As I was working on an improvement for the Compressor module, I noticed how some DSP modules have some sort of Process-method to work on a single sample and just return a single output sample. Some use ProcessBlock, some require the input by reference and work directly on the buffer while others work on a given output buffer.
This makes using the library somewhat difficult (sometimes using it inside your input loop in the AudioCallback, sometimes outside)

Now there is something to say for both of these, as processing a block reduces method calls by `blocksize` (I think? I'm somewhat new to DSP programming) but processing a single sample gives more flexibility.

I would propose adding both options to all modules, if possible. And specifically for the ProcessBlock option: never work directly on the input buffer so that stays clean should the user wish to process that input though another module.

As long as a user doesn't use both single sample- and block-processing in a single AudioCallback, it should be fine, I think.

Would there be any problem with this?

stephenhensley, 2020-07-14:
Agreed. For the most part, nearly all modules have a single-sample `Process` function, but there are certainly a few earlier additions to the library that do not (NlFilt comes to mind right away). 

It would be great to clean this up for consistency, and yeah I think its a great idea for every module to have a `ProcessBlock` function as well.

AvAars, 2020-07-14:
Allright, I'll see if I can find some time to help out on this :)