daisyaudio/DaisySP #60: Change examples to not require input signal to demonstrate functionality

andrewikenberry, 2020-07-17:
Affected modules:

- reverbsc
- pitchshifter
- stereodelay

beserge, 2020-07-17:
Pitchshift and reverb fixed via #61. examples/delayline is generated internally, perhaps stereodelay is only in DaisyExamples?

beserge, 2020-07-17:
Additionally, once these changes are okayed, they should probably be carried throughout the programmer, daisyexamples, arduino, etc.

makingsoundmachines, 2020-07-18:
maybe even keep both and rename the examples with external in pitchshifter_ext-input .. because it's the next thing people are going to code back in and it never hurts to have too many proven examples

stephenhensley, 2020-07-27:
@makingsoundmachines I like this idea. I just merged the PR that removed those examples, but time permitting we'll start adding additional examples that do use audio input as well.

This was initially brought up after a discussion of making sure that each module in DaisySP could be tested with a Daisy Seed and a single Jack wired up, without having to get find an input source or any thing extra to be able to hear what the module does. 

That said, having examples of many of these things that have audio input would be beneficial for a number of reasons. I'll open a new general issue for adding some examples that do show Audio Input as well.

stephenhensley, 2020-07-27:
Also, closing because the two examples that didn't have output-only examples have been updated.

closed 2020-07-27 by stephenhensley