daisyaudio/DaisySP #62: Faust example: include original Faust DSP code

makingsoundmachines, 2020-07-18:
Hey everyone,

suggestion for the [Faust Noise example](https://github.com/electro-smith/DaisySP/tree/master/examples/faustnoise) (and all other future Faust examples): include the Faust DSP code in a way that the Faust patch can be previewd in the [Faust IDE - faustide.grame.fr](https://faustide.grame.fr/).

Also a small tutorial how to wrap the Faust Architecture File for Daisy around which Export from the IDE (c++ code I suppose?) would be awesome!

makingsoundmachines, 2020-07-18:
There's a Faust branch for Daisy here https://github.com/grame-cncm/faust/tree/daisy
and a Faust Architecture File for Daisy lives here: https://github.com/grame-cncm/faust/tree/daisy/architecture/daisy

stephenhensley, 2020-07-27:
Yeah, this is very much underway. The existing faust example was sent to me as a proof of concept of using an early version of the faust2daisy utility. I don't think I actually have the original Faust DSP.

However, it's a great idea to have this in there as well. 

Once the Faust utility and architecture file are finalized we'll be setting up a dedicated folder in DaisyExamples with a number of FAUST examples for the various breakout boards, etc.