daisyaudio/DaisySP #64: Build of drip module fails

tvannoy, 2020-07-26:
When building, I get the following error: 
```c
modules/drip.cpp: In member function 'int daisysp::Drip::my_random(int)':
modules/drip.cpp:20:18: error: 'rand' was not declared in this scope
     return (rand() % (max + 1));
                  ^
modules/drip.cpp: In member function 'float daisysp::Drip::noise_tick()':
modules/drip.cpp:26:24: error: 'rand' was not declared in this scope
     temp = 1.0f * rand() - 1073741823.5f;
                        ^
Makefile:177: recipe for target 'build/drip.o' failed
make: *** [build/drip.o] Error 1
```

This same issue was mentioned in https://github.com/electro-smith/DaisyExamples/issues/68, but was closed by the author. The author's suggestion of adding `#include <cstdlib>` does indeed solve the problem. 

The author of the referenced issue was using Windows. I am using WSL on Windows, so I don't think this is a Windows-only issue; I can test on a proper Linux distribution to verify this.

stephenhensley, 2020-07-27:
Thanks for the heads up! We'll add the proper include so that it builds reliably.

stephenhensley, 2020-08-14:
added include to drip.cpp and tested on wsl (with old compiler) and in my normal windows environment with most recent compiler

closed 2020-08-14 by stephenhensley