daisyaudio/DaisySP #69: MIDI-CV clock script?

256k, 2020-08-17:
Hi, I was wondering if there is already a midi-CV script that could put out midi clock to trigger in order to sync my eurorack to a midi device? I'm not very familiar with C coding unfortunately so i dont know where to even begin to learn how to do it.

stephenhensley, 2020-08-18:
This is more relevant in the [libdaisy repo](https://github.com/electro-smith/libdaisy), but once MIDI clock has been added to the MIDI parser there, you could very easily send a trigger out based on the MIDI clock. 

Here's what a simple clock-to-trigger example might look like once that's been added

```C++
// global/static variables in this example
dsy_gpio trigger_output;
uint32_t time_set;
uint32_t dur = 6; // 6ms gates

// Typical Switch case for Message Type.
void HandleMidiMessage(MidiEvent m)
{
    switch(m.type)
    {
        case NoteOn:
        // do something . . .
        break;
        case Clock:
            // Sets Trigger output high, and stores the time in ms that it was set high.
            dsy_gpio_write(&trigger_output, 1);
            time_set = dsy_system_getnow();
            break;
        default: break;
    }
}
```
```cpp
// elsewhere (in callback or main)
// This shuts the trigger off after 6ms
if (dsy_system_getnow() - time_set > dur)
    dsy_gpio_write(&trigger_output, 0);
```

You could probably already do a work around where you do this on reception of a NoteOn message (using the above code, but moving it into the NoteOn section). And then just setting up a MIDI track that sends quarter notes or eighth notes for you to sync to on your eurorack.

256k, 2020-08-18:
Thank you, that makes sense the workaround you've mentioned. i could try to do that... would also be useful for transmitting the shuffle/swing of the master sequencer into euro as well coz often clocks don't transmit that shuffle.