daisyaudio/DaisySP #72: Filter consistently crashes for folded square waves

chrisfromwork, 2020-09-06:
Hello!

I am seeing a consistent filter crash when feeding in a folded square wave for a custom patch project. I've tried a variety of frequency values for the filter from 1000 - 20000 hz.

Is there any suggested tooling for setting breakpoints/debugging? I can try and track down a call stack/device state if it is possible.

chrisfromwork, 2020-09-07:
it also seems like this may be related to setting the resonance to zero. not sure if that is a supported scenario but if i set the resonance to zero + a filter frequency approaching half the sampling frequency things can crash.

stephenhensley, 2020-09-08:
@chrisfromwork Which Filter are you using? A code snippet would also be helpful for replicating the issue.

Certain filters can become unstable at frequencies lower than the nyquist, depending on resonance settings, etc. But if its crashing, especially with cutoffs as low as a 1kHz something definitely seems broken. 

There's a chance that resonance 0 is not supported, but if that's the case we can add a check since you should be able to send 0 to that setter without it crashing the filter.

chrisfromwork, 2020-09-11:
hm i have been primarily using the bandpass filter. I will need to do more testing to see if the resonance 0 is what is causing the issue. it seems a bit strange because feeding other waveforms/oscillators don't seem to cause the crash.

stephenhensley, 2020-12-14:
@chrisfromwork 

Sorry for the crazy delay on getting back to this. I went to go investigate it a bit the other day, and noticed this doesn't mention which specific Module you were using. I have seen some similar things with the `Svf` module, but just want to clarify that this is the module in question while I start digging in to see what the issue is.

chrisfromwork, 2020-12-14:
@stephenhensley 

I have not looked at this in a while, but i believe it was Svf.

For what its worth there are some other iir/fir realtime filter libraries that i've been using with daisy/patch. Might be worth looking into porting considering they're also MIT license. It took a little bit of tinkering to get past some different compiler options, but was pretty straight forward to get working.

https://github.com/berndporr/iir1
https://github.com/berndporr/fir1

stephenhensley, 2020-12-14:
@chrisfromwork That sounds about right. I've seen a few other instabilities with that filter lately. I'll try to get it fixed up in the next few weeks.

And awesome! thanks for sharing these, I hadn't seen these particular repos before. We'll definitely start looking into how they'll fit in with our current library structure! Always looking for more stuff to add 😄

chrisfromwork, 2020-12-14:
They're pretty slick. I personally wouldn't want to be responsible for writing an iir implementation of anything. :P

stephenhensley, 2021-01-29:
So I've tested #126 with a 25Hz square wave (naive, not bandlimited), and async phasers for each of the parameters to try to get it to crash (it was crashing quite easily when I started).

One limitation of the filter has changed, as I was unable to get it to be reliably stable above one third of the frequency range. So instead of the max cutoff being half of the samplerate it is now one third of the samplerate.

closed 2021-01-29 by stephenhensley