daisyaudio/DaisySP #78: Adenv crashes with segment times = 0

andrewikenberry, 2020-10-28:
As dicussed [on the Daisy forum](https://forum.electro-smith.com/t/adenv-crashes-with-segment-times-0/800)

andrewikenberry, 2020-10-28:
[image] [Fix AdEnv non-zero issue](https://trello.com/c/eBR0nocx/32-fix-adenv-non-zero-issue)

sdiedrichsen, 2021-06-08:
This should have been fixed with my remake, which has been merge recently.

makingsoundmachines, 2021-07-26:
I'm having some issues with the ADenv as it is right now: 
- with long release times, a decay tied to a pot (via parameter) will sometimes get "stuck" and keep passing the audio through; 
- with very short times close to 0.0 (e.g. 0.0001) and other DSP load going on, I get audio glitches that are gone at more moderate, "middle of the road" segment times

A suggestion (as discussed in the electrosmith slack) would be to apply @sdiedrichsen 's ADSR fix to the ADenv as well: https://github.com/electro-smith/DaisySP/pull/160

makingsoundmachines, 2021-07-26:
A lower limit for segment values of 0.0005 (for ADENV_SEG_ATTACK / ADENV_SEG_DECAY) fixed the issues I was having.