daisyaudio/DaisySP #86: Phasor's name ?

MrBlueXav, 2020-12-22:
I'd like to know why this module is called "phasor".
It seems to be just a ramp generator or sawtooth oscillator and not a complete phasor effect.
Thank you for clarifying that.
Xavier

andrewikenberry, 2020-12-22:
A [phasor](https://electronicmusic.fandom.com/wiki/Phasor) is a specific type of signal used for incrementing phase for oscillators, table lookup, sample playback position etc.

It is not to be confused with the common phaser effect.

Hope this helps!

closed 2020-12-22 by andrewikenberry

MrBlueXav, 2020-12-23:
OK thanks !

PaulBatchelor, 2021-01-18:
I did some digging into where "phasor" gets its name, because I really didn't know. It's resemblance to "phaser" always confused me.

In the physics/engineering, the term "phasor" (portamenteau of phase vector) is used to describe a complex value of a sinusoid (it's phase, and amplitude). What we think of as a "phasor" in the computer music context could be construed to be just the "phase" part of the phasor, normalized to be in range 0-1. This sort of makes sense when you think that this is traditionally used to supply the phase position in a table-lookup oscillator.

Speaking of tradition, using the term "phasor" to describe this kind of normalized periodic ramp signal probably originated in Csound, or maybe even earlier in MUSICN languages. It's a very very old Csound opcode. I can't help but notice that "phasor" is six letters long, which at one point was the maximum length of opcode name. So, overloading terminology aside, "phasor" is a pretty decent 6-letter description for what it does.

Some more links:

[phasor on wikipedia](https://en.wikipedia.org/wiki/Phasor)

[phasors from Mathematics of the DFT by Julius Smith](https://ccrma.stanford.edu/~jos/st/Phasor.html)

MrBlueXav, 2021-01-19:
That's really a great explanation ! Thank you very much !
In fact I didn't notice the difference in the spelling of phas**or** and phas**er**... Sorry about that !