daisyaudio/DaisySP #134: Bitcrush does not work properly

jerry20091103, 2021-02-04:
The bitcrush module is outputing a huge DC offset in my testing, it seems that there's a bug in the code.
In line 27 of bitcrush.cpp:
`out *= (65536.0f / bits) - 32768;`
I believe it should be
`out *= (65536.0f / bits);`
`out -= 32768;`
or something equivalent instead.

stephenhensley, 2021-02-04:
Thanks for pointing this out. 

I was just discussing with someone this morning how we have multiple modules that accomplish similar goals, and we may remove a few to consolidate all downsampling/bitcrushing to a single module.

For now, you can try the Decimator module as an alternative.

PaulBatchelor, 2021-03-31:
Another issue with BitCrush is that the internal `fold` transformation will eventually suffer from numerical overflow (both integer AND floating point), which is UB. Reason for this is that `fold` uses two indefinitely incrementing counters to keep track of absolute time. It's also been a longstanding bug in Soundpipe all this time. Oops.