daisyaudio/DaisySP #136: Overriding dsp.h

antisvin, 2021-02-05:
There are a few problems with current implementation that I'd like to solve with a different version of functions defined in dsp.h:

1. current fmin/fmax code for ARM doesn't work with M4 MCUs, because it uses inline ASM that only works with VFP5 FPU on M7.
2. some functions could use platform specific implementations that are already available (i.e. pow/log LUTs on OWL)

Currently I'd have to create a forked version of DaisySP to replace dsp.h for custom implementation.

I wonder if it's not unreasonable to ask for such replacement to be supported officially. The problem with current code is that with quoted includes there's no way to shadow that header with a different file. Something that could probably work is including it like this:
```
#ifndef DAYSYSP_OVERRIDE_DSP_H
#include "dsp.h"
#else
#include <custom_dsp.h>
#endif
```
I can provide a PR for this or maybe some better approach could be used for overriding this file.

antisvin, 2021-02-05:
Ok, a better approach would probably be to handle importing custom header from dsp.h itself. I.e. by using -DDSY_CORE_DSP -D DSY_OVERRIDE_DSP when compiling, then checking for second flag's presence inside dsp.h and including the extra file in such case.

stephenhensley, 2021-02-09:
This is a really interesting point, and it first came up with the inclusion of the inline asm for the fmin/fmax. I had overlooked the fact that many other ARM processors would have different asm for that.

I like the compile time definition option the most of these, two.

I'd also be open to splitting "dsp.h" up into a few files: one with the math implementations that could be overwritten with custom code, and one with platform independent utility stuff (mtof, median filter, etc.). 

That said there aren't a lot of utility functions yet, but if you're overriding math functions you might still want some of the other stuff around.

antisvin, 2021-02-09:
I think that the main issue is that this file can't be replaced other than by forking library. Having non-platform specific utilities in a separate header would be nice, but it can happen later and would only reduce amount of copy-pasted code in the custom version.

stephenhensley, 2021-02-09:
True, good point. Longterm we can split it to "dsp.h" -- the file that can be replaced with the define, and something like "core.h" with just agnostic helper functions.

For now I'm down with taking your approach and requiring some copy-paste if you want to keep the utility functions.

stephenhensley, 2021-02-09:
Closing since the relevant PR is merged.

closed 2021-02-09 by stephenhensley