daisyaudio/DaisySP #137: String.h breaking compilation after directory refactoring

antisvin, 2021-02-05:
New directory structure requires listing all folders with -I parameter to gcc. As a result, string.h ends up shadowing system header when something like `#include <string.h>` is used in third party code

It was not an issue previously, as only top level directory was added to include list.

antisvin, 2021-02-05:
Managed to solve this by using "-idirafter" instead of "-I" for including DaisySP directories in custom makefile used to build it. Feel free to close this issue, unless someone decides that this header needs a better name.

zeroisnan, 2021-02-07:
I came here to report the same issue, I worked around using -iquote (see gcc manuale [here](https://gcc.gnu.org/onlinedocs/gcc/Directory-Options.html)).

Regardless of the specific gcc option used, I believe this is an issue that should be addressed at the library level (by renaming the file) rather than in the build system, as you do have control over the code, while have no control on how users will integrate the library in their projects.

stephenhensley, 2021-02-09:
Ha, wow. Somehow three of us looked at that module name and didn't think, "That's going to cause problems down the road".

I'd even just be open to changing the name to avoid a clash with the keyword in general.

stephenhensley, 2021-02-09:
I'll have a PR in a few minutes, and have renamed the filename to `KarplusString.h` instead to make it a bit clearer what it is, and to avoid the name conflict.

closed 2021-02-09 by stephenhensley