daisyaudio/DaisySP #149: Building on a Raspberry Pi ?

madskjeldgaard, 2021-03-18:
Hello all

Thanks for this great library. I have implemented some of the classes from DaisySP in a little collection of SuperCollider plugins (https://github.com/madskjeldgaard/mkplugins) and it works really great! Thanks so much... 

The only problem I have found is that I can't seem to build the Daisy library on a  Raspberry Pi. 
 
I am trying on a Raspberry Pi 4 armv7l running cmake version 3.16.3.

Here's the error:

```
Scanning dependencies of target DaisySP
[  1%] Building CXX object DaisySP/CMakeFiles/DaisySP.dir/Source/Control/adenv.cpp.o
[  2%] Building CXX object DaisySP/CMakeFiles/DaisySP.dir/Source/Control/adsr.cpp.o
[  3%] Building CXX object DaisySP/CMakeFiles/DaisySP.dir/Source/Control/line.cpp.o
[  4%] Building CXX object DaisySP/CMakeFiles/DaisySP.dir/Source/Control/phasor.cpp.o
[  5%] Building CXX object DaisySP/CMakeFiles/DaisySP.dir/Source/Drums/analogbassdrum.cpp.o
/tmp/ccCj9Fuc.s: Assembler messages:
/tmp/ccCj9Fuc.s:65: Error: selected processor does not support `vminnm.f32 s20,s20,s15' in ARM mode
/tmp/ccCj9Fuc.s:179: Error: selected processor does not support `vmaxnm.f32 s0,s0,s16' in ARM mode
/tmp/ccCj9Fuc.s:182: Error: selected processor does not support `vminnm.f32 s16,s0,s22' in ARM mode
/tmp/ccCj9Fuc.s:355: Error: selected processor does not support `vmaxnm.f32 s0,s0,s15' in ARM mode
/tmp/ccCj9Fuc.s:358: Error: selected processor does not support `vminnm.f32 s0,s0,s14' in ARM mode
/tmp/ccCj9Fuc.s:390: Error: selected processor does not support `vmaxnm.f32 s0,s0,s14' in ARM mode
/tmp/ccCj9Fuc.s:393: Error: selected processor does not support `vminnm.f32 s0,s0,s15' in ARM mode
/tmp/ccCj9Fuc.s:423: Error: selected processor does not support `vmaxnm.f32 s0,s0,s15' in ARM mode
/tmp/ccCj9Fuc.s:426: Error: selected processor does not support `vminnm.f32 s0,s0,s14' in ARM mode
make[2]: *** [DaisySP/CMakeFiles/DaisySP.dir/build.make:115: DaisySP/CMakeFiles/DaisySP.dir/Source/Drums/analogbassdrum.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1022: DaisySP/CMakeFiles/DaisySP.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

stephenhensley, 2021-03-18:
Woah, cool to see it making it's way to both RPi and SuperCollider. It looks like we'll to swap out the `__arm__` define check for something more specific to the M7 FPU. I'll check for a more appropriate flag, and replace it ASAP.

In the meantime, if you want to confirm the rest of the build, you can temporarily comment out, or delete lines 35-37 and 46-48 in `Source/Utility/dsp.h`

If you hit any other snags feel free to just ping me back on this issue. I'll chime back in and close this once I've resolved this issue. I have a few Pis lying around here. So I should be able to do a quick test on my end as well.

madskjeldgaard, 2021-03-19:
Yep - removing the preprocessor lines for `__arm__` definately did the trick. Works fine on my Raspberry Pi 4 here now

madskjeldgaard, 2021-03-19:
PS. What attracts people to using the SuperCollider plugins on the Pi is for many to use it in the [monome norns](https://github.com/monome/norns/) eco-system so you should see some DaisySP on there when this is fixed.

And also thanks again for the nice work. This library is wonderfully well structured and easy to use. If you need a reference later on for people wanting to use it with supercollider plugins, this section of my CMakeLists.txt is what I use to link each plugin to the library https://github.com/madskjeldgaard/mkplugins/blob/bc006ae4c3718df8ef7aa668bd1d2eb18f7db5b8/CMakeLists.txt#L138

stephenhensley, 2021-03-19:
Awesome! Thanks for checking, and thanks for the additional info!

I should be able to get around to merging a patch to resolve this within the next week.