daisyaudio/DaisySP #150: Shaper oscillators emit DC

madskjeldgaard, 2021-03-31:
Hello

Something I just noticed when using DaisySP in my SuperCollider plugins is that some of the shaper oscillators (ZOsc, VarShape and varSaw) emit DC and it is necessary to use a dc blocker in between their outputs and the system output at the moment. For ZOsc and VarShape it depends on the "shape" parameter it seems how much dc is in there. 

ThankS!

madskjeldgaard, 2021-03-31:
These are the affected oscillators :  
  - formant
 - variablesawosc
  - variableshapeosc 
 - vosim
 - Zoscillator

sdiedrichsen, 2021-06-08:
I think, that behaves correctly. E.g. take a classic Pulse oscillator with PWM. Only at 50% it's free of DC, at any other pulse width, it has a DC offset. But I never saw an analog implementation removing it by a feedforward of the PW voltage to the oscillator. So, inserting a high pass filter is an requirement.