daisyaudio/DaisySP #154: ADSR produces small step and audible click in decay or release phase

sdiedrichsen, 2021-05-03:
https://es-daisy.slack.com/archives/CU7QB1K5Y/p1620032978070300

There's a rather unfortunate comparison, which forces the output to zero once the envelope falls below 0.01.

I'd suggest using a negative target value for the the filter and clip the output at zero. Also, for the attack a target value > 1.0 produces a more pleasant attack phase.

jasonmayo, 2021-05-03:
There are situations where it would be useful to set ADSR_SEG_IDLE at a given level, such as prioritising voices for polyphonic voice assignment. Could the current 0.01 be a default that can be overridden?
`if(out <= idle_threshold)`

sdiedrichsen, 2021-05-03:
I'd tend to disagree a bit, the ADSR should produce a clean envelope primarily. But for voice allocation or stealing a accessor method for the segment might be useful.

sdiedrichsen, 2021-05-17:
OK, here's my remake:
https://github.com/electro-smith/DaisySP/pull/160

sdiedrichsen, 2021-06-08:
This has been merged, so this issue can be closed now.

closed 2021-06-08 by sdiedrichsen