daisyaudio/DaisySP #156: C++ 11 compatibility

antisvin, 2021-05-09:
[This function](https://github.com/electro-smith/DaisySP/blob/master/Source/Utility/dsp.h#L270-L282) is raising the following error if DaisySP is being built for C++11:

```
 error: body of constexpr function 'constexpr uint32_t daisysp::get_next_power2(uint32_t)' not a return-statement
```

That's because that version of C++ standard mandates constexpr to include just a single return statement, later revisions lifted that restriction. The function itself is not used anywhere in library code, but it was added as part of FIR filter commit.

So it should probably be removed, rewritten or guarded by __cplusplus version check if compatibility with C++11 is desired.

TheSlowGrowth, 2021-05-09:
Do you actually need C++11? It's pretty ancient... Sure. Things can be refactored, but why?

antisvin, 2021-05-09:
I would be fine using either version, but C++11 is used by existing build system. So I'm not really making a choice in this case.

I've opened this issue because recent commit broke compatibility due to including code that is not even used by library itself. This looks like an unintended regression. So this can be either fixed in library or just by using custom dsp.h implementation that doesn't include those unused functions.

stephenhensley, 2021-05-12:
While not currently used, I do see the value of a helper function like this.

I'm in favor of either guarding it by a check for the c++ version, or making it a standard function instead of constexpr on versions less than c++14. Something like:

```c++
#if __cplusplus <= 201103L
uint32_t get_next_power2(uint32_t x)
#else
constexpr uint32_t get_next_power2(uint32_t x)
#endif
{
    // function body
    return x;
}
```

It is a bit messy looking, and leads to different behavior on different c++ versions, but it might be a good solution. If we start doing this more than just here, we should probably make a specific macro for that check.

I can make a PR if everyone likes this idea. I'm also open to other ideas.

antisvin, 2021-05-12:
I don't see any problems with this approach, this difference in behavior won't matter since the functions are not used by library. Maybe add a short comment explaining why it's done like that.

As a solution to more general problem of supporting multiple C++ versions, maybe CI could be configured to make several runs targeting each C++ standard version that would be considered supported? Such list of supported versions doesn't even seem to be even documented/stated anywhere.

polyclash, 2021-05-12:
costexpr have copile time optimization benefits if its input isn't const?

TheSlowGrowth, 2021-05-12:
@stephenhensley The code could be much tidier like this:
```cpp
// somewhere in platform.h
#if __cplusplus <= 201103L
#   define D_CONSTEXPR constexpr
#else
#   define D_CONSTEXPR
#endif

// elswhere in daisySP
#include "platform.h"
D_CONSTEXPR uint32_t get_next_power2(uint32_t x)
{
    // function body
    return x;
}
```

antisvin, 2021-05-12:
Introducing a separate macro for the sake of this looks a bit excessive to me.

stephenhensley, 2021-05-13:
Yeah, I think if this is an exception, and not a general rule we can leave it semi-ugly to make it a point that it's not idiomatic with the rest of the library. However, if this becomes a trend with other helpers we can add any necessary preprocessors.

I just opened a PR for this #159 -- I'll give it a quick compile/test on multiple on c++11 in the morning. So feel free to weigh in if anyone thinks anything should change.

PR #159 (2021-05-19): Adds guard around constexpr function to allow C++11 compatibility
fixes #156 - added guard around constexpr to prevent issues with c++11 compatibility.
commit bb80c4c1 (2021-05-19): Adds guard around constexpr function to allow C++11 compatibility (#159)

* fixed #156 - added guard around constexpr to prevent issues with c++11 compatibility.

* inlined the function when it's not constexpr to prevent multiple definition error.
closed 2021-05-19 by stephenhensley