daisyaudio/DaisySP #169: std::sin usage in Looper

antisvin, 2021-11-18:
I've ran into an issue with the latest library version due to std::sin being used to [define a constant in Looper.h](https://github.com/electro-smith/DaisySP/blob/master/Source/Utility/looper.h#L226) . The problem is that on OWL we replace sin/sinf with calls to arm_sin_f32 from CMSIS by overriding it with a macro. This won't work when it's used from std namespace and it can't be constexpr because of this optimization.

Would it be possible to either use the actual value of 0.7071067811865476  for looper or move it to dsp.h to allow overriding in third party code?

stephenhensley, 2021-11-19:
Yup! Easy enough; thanks for the feedback!

antisvin, 2021-11-19:
Thanks, this should do the trick.

stephenhensley, 2021-11-19:
no problem!

PR #170 (2021-11-19): Looper - Updated Frip Decay value to use explicit value instead of sin constexpr
fixes #169 - change constant value for frip decay from std::sin to fixed value.
commit 4263388e (2021-11-19): Looper - Updated Frip Decay value to use explicit value instead of sin constexpr (#170)

* fixes #169 - change constant value for frip decay from std::sin to fixed value.

* style fix.
closed 2021-11-19 by stephenhensley