daisyaudio/DaisySP #171: Svf filter does not like to be modulated

madskjeldgaard, 2021-12-09:
Hello friends

I was messing around with the Svf filter today inside of a delay's feedback loop on my Daisy Patch and it sounded really great when not modulated, but when applying modulation, it stops the sound at some point. It sounds a bit like a NaN (and I tried adding dc filters to avoid this, but to no avail). When switching out the DaisySP Svf class with the Svf filter used in Mutable Instruments' modules, it worked fine in the same usecase.

Best

stephenhensley, 2021-12-09:
@madskjeldgaard thanks for the bug report. We used to have a few NaN issues, that I thought had gotten resolved (some ratios between drive/resonance made the filter susceptible to instability).

Also, if it is a NaN issue, it's usually non-recoverable since it gets into the feedback path. Was sound cutting out and coming back, or did you have to reset the device/power-cycle when it cut out?

Can you recall what modulation you were sending it? Perhaps a negative frequency value or something?

Thanks!

TheSlowGrowth, 2021-12-10:
> Perhaps a negative frequency value or something?

That's a good point. Please make sure that you are properly clamping the parameters to their legal ranges. If you haven't done that yet, could you please add the clamping and try the same wild modulation again? Does the problem persist?

madskjeldgaard, 2021-12-17:
sorry for the late reply. 

I am not 100% sure if there were negative values or not.

The parameter was scaled like this:

```cpp
params[2].Init(patch.controls[2], 20, 20000, Parameter::LOGARITHMIC);
```
and then inside of my processing function it was filtered using a onepole filter

```
// Set cutoff
fonepole(fbcutoff, cutoffTarget, .0002f);
svf->setFrequency(fbcutoff);
```
I had to restart the daisy to revive it, I think.