daisyaudio/DaisySP #172: CMake errors

FrantisekPostl, 2021-12-23:
Hi, I'm trying to build daisysp on Mac and I'm getting these errors:

```
Consolidate compiler generated dependencies of target DaisySP
[  1%] Building CXX object CMakeFiles/DaisySP.dir/Source/Control/adsr.cpp.o
In file included from /Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.cpp:1:
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:82:13: error: function definition does not declare parameters
    float   sus_level_{0.f};
            ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:83:13: error: function definition does not declare parameters
    float   x_{0.f};
            ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:84:13: error: function definition does not declare parameters
    float   attackShape_{-1.f};
            ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:85:13: error: function definition does not declare parameters
    float   attackTarget_{0.0f};
            ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:86:13: error: function definition does not declare parameters
    float   attackTime_{-1.0f};
            ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:87:13: error: function definition does not declare parameters
    float   decayTime_{-1.0f};
            ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:88:13: error: function definition does not declare parameters
    float   releaseTime_{-1.0f};
            ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:89:13: error: function definition does not declare parameters
    float   attackD0_{0.f};
            ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:90:13: error: function definition does not declare parameters
    float   decayD0_{0.f};
            ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:91:13: error: function definition does not declare parameters
    float   releaseD0_{0.f};
            ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:93:13: error: function definition does not declare parameters
    uint8_t mode_{ADSR_SEG_IDLE};
            ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:94:13: error: function definition does not declare parameters
    bool    gate_{false};
            ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:70:9: error: use of undeclared identifier 'sus_level_'; did you mean 'sus_level'?
        sus_level_ = sus_level;
        ^~~~~~~~~~
        sus_level
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:66:39: note: 'sus_level' declared here
    inline void SetSustainLevel(float sus_level)
                                      ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:75:49: error: use of undeclared identifier 'mode_'
    inline uint8_t GetCurrentSegment() { return mode_; }
                                                ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.h:79:44: error: use of undeclared identifier 'mode_'
    inline bool IsRunning() const { return mode_ != ADSR_SEG_IDLE; }
                                           ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.cpp:10:5: error: use of undeclared identifier 'attackShape_'
    attackShape_  = -1.f;
    ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.cpp:11:5: error: use of undeclared identifier 'attackTarget_'
    attackTarget_ = 0.0f;
    ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.cpp:12:5: error: use of undeclared identifier 'attackTime_'
    attackTime_   = -1.f;
    ^
/Users/frantisekpostl/Documents/DaisySP/Source/Control/adsr.cpp:13:5: error: use of undeclared identifier 'decayTime_'
    decayTime_    = -1.f;
    ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [CMakeFiles/DaisySP.dir/Source/Control/adsr.cpp.o] Error 1
make[1]: *** [CMakeFiles/DaisySP.dir/all] Error 2
make: *** [all] Error 2
```

software versions:
macos 12.0.1
cmake 3.22.1
compiler Apple clang version 13.0.0

do you need some additional information? I'm new to c++ and cmake.

stephenhensley, 2022-01-03:
Hi @FrantisekPostl thanks for the thorough error report.

Sorry for the delay, many of us at electrosmith were out of the office for the holidays.

I'll try to recreate this on MacOS some time this week, and let you know if I have any ideas for fixing it.

I haven't tried it in a while, but about a year ago we set up a sample project for using DaisySP and JUCE. I haven't tested it on Mac OS yet, but it might be a good jumping off point if you're new to both C++ and CMake as it was a standalone project that would make sound instead of just the library on it's own. (Alternatively, I do plan on putting together a similar basic project for DaisySP + Portaudio in the near future as well).

In the meantime, while we attempt to recreate the issue, if you could provide some additional info it would be helpful:

* Are you including DaisySP in another program, or are you simply just trying to compile the library?
* Is your Mac running an intel processor or the new M1 architecture?

Thanks!

FrantisekPostl, 2022-01-04:
Hi, i didn't expected answer sooner than after new year. I've compiled it with gcc and it's working. I don't know why but at least some clue. I have intel processor and i was trying just compile library.

takumi-ogata, 2023-02-17:
It looks like the issue was resolved by compiling with gcc. Please feel free to reopen if it's still a problem.

closed 2023-02-17 by takumi-ogata

alexander-daniel, 2023-12-07:
I'm very sorry to ask a silly question, but how do I change it so I can compile with gcc? I'm trying to run the examples in the DaisySP + JUCE example