daisyaudio/DaisySP #173: moogladder tanh incorrect

Sin-tel, 2022-07-22:
I was just looking at the code of the moogladder out of curiosity, and noticed that the fast tanh implementation seems fairly broken.

https://github.com/electro-smith/DaisySP/blob/4263388e7fa8dfd34fa85c6a1c697362dc6981c7/Source/Filters/moogladder.cpp#L6-L24
The referenced implementation in soundpipe is:

https://github.com/adiog/soundpipe/blob/35d856fcf8a0d85c285818554f569b592d066d10/modules/moogladder.c#L26-L42
Which is clearly not the same thing at all!

Simple correction:
```cpp
float MoogLadder::my_tanh(float x)
{
    int sign = 1;
    if (x < 0) { 
        sign = -1; 
        x = -x;
    }
    if (x >= 4.0) {
      return sign;
    }
    if (x < 0.5) return x * sign;
    return sign * tanh(x);
}
```

I did not test any of this, so maybe I'm wrong.

takumi-ogata, 2023-02-17:
Thank you for noticing this! Please feel free to test and submit a PR. We would greatly appreciate it. Thanks!

beserge, 2023-12-20:
Nice find! Tested and fixed. This will be a part of the upcoming [LGPL split](https://github.com/electro-smith/DaisySP-LGPL)

closed 2023-12-20 by beserge