daisyaudio/DaisySP #195: Wrong code in biquad.cpp

tobybear, 2023-11-20:
There is a bug/typo in the biquad init function in biquad.cpp.

Line 12 currently is this:
float beta  = 1.0f + cosf(con);
While it should be this:
float beta = res_ * res_ * sinf(2 * con) - 2 * _res * cosf(con) * sinf(con);

This seems to be a copy and paste error, since the original source where the code came from (the Soundpipe library) does have the correct line in place, and besides, the expression assigned to variable beta is actually the same as the one assigned to variable gamma in the next line.
Compare lines 45ff from here for verification:
https://github.com/PaulBatchelor/Soundpipe/blob/master/modules/biquad.c

takumi-ogata, 2023-11-21:
Hi tobybear,

Thank you so much for bring this up!
If you would like to, you're more than welcome to make that change and submit a PR. We would greatly appreciate it. Otherwise, we will address this when we get a chance.

beserge, 2023-12-20:
Thanks so much for the fix! This will be addressed in the upcoming [LGPL split](https://github.com/electro-smith/DaisySP-LGPL)

closed 2023-12-20 by beserge

satyrVTteam, 2025-01-02:
Even after the fixed line this code doesn't work.
I replaced it with this one and it worked from the start https://www.earlevel.com/main/2012/11/26/biquad-c-source-code/
It is easy to understand and easy to work with.

tobybear, 2025-01-02:
Yes, the original filter code is definitely broken, best to use a better implementation, e.g. like the one @satyrVTteam suggested above!