daisyaudio/DaisySP #199: [REQ] fix license on GH

forart, 2024-01-09:
Hi there, 1st of all thanks for your cool work !

Since we've "doxed" it in our [HyMPS project](https://github.com/forart/HyMPS#readme) (under the [AUDIO](https://github.com/forart/HyMPS#-) \ [Libraries](https://github.com/forart/HyMPS/blob/main/VideoTools.md#--) \ [DSPs subsection](https://github.com/forart/HyMPS/blob/main/AudioLIBs.md#dsps-)), can you please "standardize" the license for it ?

Although it may sounds like a minor aspect, a GH "uncompliant" license file causes an inconsistent generation of the relative badge:

[image]
(badge-generator URL: https://flat.badgen.net/github/license/electro-smith/DaisySP/?label=LICENSE)

You can easily set a standardized license file through the GH's license wizard tool.

Last but not least, let us know how - in your opinion - we could improve our categorizations and links to resources in order to favor collaboration between developers (and therefore evolution) of listed projects.

Thanks in advance.

beserge, 2024-02-01:
Hi @forart, thanks for including us in your repository, we're really honored!
For now, we unfortunately can't change the license file, but I was able to find a way with badgen to statically create the badge
to display the correct license. Of course this isn't as clean of a solution as dynamically generating them, but we don't plan on moving from the MIT license any time soon!

Here's the url, as well as how it looks rendered.
https://flat.badgen.net/static/license/MIT/blue?label=
[image]

Thanks for your feedback!

closed 2024-02-01 by beserge

forart, 2024-02-02:
Thanks for reply.

> Here's the url, as well as how it looks rendered. https://flat.badgen.net/static/license/MIT/blue?label= 
> [image]
> 
Yes we know (we use this approach for multi or own licensing), but we would like to avoid it for GH's officially supported licenses.

Thanks anyway.