daisyaudio/DaisySP #206: negative semitones issue

evilcomputer12, 2024-11-02:
hi, i still have an issue with setTranspose(-semitones) for shifting semitones up it works bit for shifting semitones down audio is muted, if its fixed maybe i am using it wrong, can someone show an example how to shift semitones down

xconverge, 2024-11-11:
Use the code from here https://github.com/electro-smith/DaisySP/pull/166

evilcomputer12, 2024-11-11:
i've tried that but when i put -1 it goes mute for +2 for example it works

xconverge, 2024-11-11:
> i've tried that but when i put -1 it goes mute for +2 for example it works

We briefly crossed paths on reddit a few weeks ago by the way!

Here is my implementation: https://github.com/bkshepherd/DaisySeedProjects/pull/24

I am using the code referenced in #166  successfully for both negative and positive semitones. Do you have any code I can peek at to try and help further? I have made some changes since related to buffer sizes scaling linearly the more semitones you go but that PR was certainly functional

evilcomputer12, 2024-11-20:
i've just did setTranspose(-1)